FASTSHIFT/X-TRACK #6: 增加对蓝牙，ant+设备的支持，然后再增加strava live segment，然后续航再长一点就完美了

Pluto-leek, 2021-07-19:


Faultiness, 2021-07-19:
[image]

FASTSHIFT, 2021-07-19:
@Pluto-leek 
ANT+协议不开源，如果要使用，每年要向佳明交几千美金授权费用。
strava live segment应该是佳明的收费服务。要浏览轨迹可以用免费的GPXSee。

liuxin27, 2021-07-22:
> @Pluto-leek 
> 
> ANT+协议不开源，如果要使用，每年要向佳明交几千美金授权费用。
> 
> strava live segment应该是佳明的收费服务。要浏览轨迹可以用免费的GPXSee。
增加蓝牙是不是比较容易实现，迈金的速度踏频和心率带都支持双模

FASTSHIFT, 2021-07-22:
> > @Pluto-leek
> > ANT+协议不开源，如果要使用，每年要向佳明交几千美金授权费用。
> > strava live segment应该是佳明的收费服务。要浏览轨迹可以用免费的GPXSee。
> > 增加蓝牙是不是比较容易实现，迈金的速度踏频和心率带都支持双模

走BLE应该可以

arinport, 2021-07-23:
@FASTSHIFT 

按照Garmin的官方说法，是你使用ANT+的商标，或者声称支持ANT+才需要缴纳费用。我们这个属于爱好者项目，犯不着：
https://www.thisisant.com/developer/ant-plus/certification-faq/category/258/

> No.  For authorization to apply ANT+ brands, logos or wordmarks to any product  or claim ANT+ compatibility in any way,  all implemented profiles must be certified. 

我们并没有**最终做出来**一个产品，只是给一个图纸，至于兼容不兼容，你拿来做产品才需要考虑。正如Linux严格上来说不是UNIX，但没人否认一样。

第二，ANT+的芯片其实蛮多的：https://www.mouser.sg/Search/Refine?Keyword=nRF51422

https://www.nordicsemi.com/-/media/Software-and-other-downloads/Product-Briefs/nRF51422-product-brief.pdf?la=en&hash=254A7F877BB992E90C09329D5B954CD7770B9016

我个人认为不是太大的问题。

第三，我认为考虑现在芯片大涨价的背景下，一些不那么主流的芯片可能可以考虑了，例如V3s，f1c100s，还有大热的K210。甚至于，可以开启省电模式，有省电功能屏幕或者UI只运行上面那个集成了Cortex-M0的ANT+芯片，以取得最佳的续航。

我个人会往这个方面考虑

FASTSHIFT, 2021-07-23:
@arinpo
不错的提议，后期会考虑的

FASTSHIFT, 2021-07-23:
> @FASTSHIFT
>
> 按照Garmin的官方说法，是你使用ANT+的商标，或者声称支持ANT+才需要缴纳费用。我们这个属于爱好者项目，犯不着：
> https://www.thisisant.com/developer/ant-plus/certification-faq/category/258/ [...]

不错的提议，后期会考虑的

zhangchunzhong, 2021-07-25:
附议。加入功率计，速度计，心率计，这样才比较专业。现在这些外设都是ant和蓝牙双协议的。如果ant不行，就只支持蓝牙就行。
另外，参考佳明的设计，显示页面可配置也非常重要。

zhangchunzhong, 2021-07-25:
如果你们加入蓝牙，我就参加开发。。。

arinport, 2021-07-26:
@zhangchunzhong 

得先选型啥芯片吧。要考虑产量/供应链/生产难度/温湿度要求/功耗一系列问题。不然写了代码也是白写

ANT+我在官网只看到了nRF系列芯片支持，同时支持蓝牙。但是现在的供应链不是那么稳定。另外STM32现在的价格已经疯了，已经比很多其他类型的芯片要贵上10倍了。

至于写代码，这个反而是最稳定没啥疑虑的问题了。v3s/f1c100s都够上Linux了。

arinport, 2021-07-26:
@zhangchunzhong 

https://www.nordicsemi.com/-/media/Software-and-other-downloads/Product-Briefs/nRF52832-product-brief.pdf

你看下这个，是Cortex-M4的CPU，硬件规格和 @FASTSHIFT 的版本差不多一致，多了蓝牙和ANT+的功能。某宝有SDK买（搜nRF52），不需要你做板子。

期待你能连上Garmin的功率计和速度计

[image]

FASTSHIFT, 2021-07-26:
@arinport 
在当前体验不打折扣的情况下，MCU的性能：
主频 >= 200MHz（如果有GPU，主频要求可以降低，但不能低于100MHz）
RAM >= 200K（至少要有完整的帧缓冲保证显示不撕裂，lvgl需要至少32K~64K的内存池，保证完整的轨迹绘制的功能）
FLASH >= 512K （当前代码尺寸大约260KB）
之前也关注过这个芯片，但是受限于性能（应该是和功耗做了妥协），没有继续研究。

您会使用这个蓝牙芯片吗，欢迎向这个项目提交PR，完善蓝牙功能。

arinport, 2021-07-26:
@FASTSHIFT 

价格上大概率会用便宜一点的。这个芯片太贵了。

我初步的构想会是

1. F1c100s/F1c200s 400MHz - 1.1GHz (集成32MiB/64MiB RAM)， 电流可能在100mA左右 （之所以电流低是因为28nm的工艺）
2.一个16MiB/32MiB的Flash
3. nRF51422， 电流约 10mA - 20mA，低功耗模式下只开这个MCU

_**但是这样子可能改动太大了**_。

硬件项目还是要您先沟通好。例如这个芯片的产能，板子的面积，功耗，散热等等问题。

FASTSHIFT, 2021-07-26:
@arinport 
那您对F1c100s/F1c200s开发熟吗？倾向于上linux还是用裸机开发？
我之前也玩过这个芯片，但是开发资料太碎了，学习成本不低。

arinport, 2021-07-26:
@FASTSHIFT 

我目前只试过Linux和RT-Thread。资料非常少。好处是Linux确实是可以跑起来的，速度还不算慢。我正式项目用过C++和Golang（GCC-go）的程序。不过我没做过您那么复杂的UI

arinport, 2021-07-26:
@FASTSHIFT 

关键在于第三点nRF51422能否跑起来，连接ANT+设备（例如4个），连接以后的电量消耗，以及是否能带得动低功耗模式下的UI。

至于是不是f1c100s，换个其他芯片区别也不是很大。

FASTSHIFT, 2021-07-26:
@arinport 
关于ANT+，如果就算能做出来，我不确定佳明是否允许开源，商业化的东西需要谨慎。

arinport, 2021-07-26:
@FASTSHIFT

做一个github repo不至于吧。Nordic的SDK可以随便下载。至于真的做成产品，我感觉更大的问题是有没有人买的问题。行者冲进去结果发现销量实在太小了。被Garmin追究反而是一个幸福的烦恼。

坦白说，最佳商业化路线应该是HUD，目前的汽车有HUD寥寥无几，选配费用很高，而且大部分HUD做得很烂。如果您真能做到GPS测速+导航+阳光下看清屏幕，做HUD岂不美哉？

Pluto-leek, 2021-07-27:
> @arinport
> 关于ANT+，如果就算能做出来，我不确定佳明是否允许开源，商业化的东西需要谨慎。

ANT+不支持，支持蓝牙也可以，现在自行车行业的主流感应器基本都是蓝牙，ant+双模的

doyou, 2021-07-27:
是不是可以建议采用e-ink屏幕以保障续航及高亮下清晰？

FASTSHIFT, 2021-07-28:
有谁会写App吗？或者有BLE开发经验?

FASTSHIFT, 2021-07-28:
> 是不是可以建议采用e-ink屏幕以保障续航及高亮下清晰？

墨水屏刷新体验不好，而且不适合在阳光下暴晒。

zhangchunzhong, 2021-07-28:
> 有会写App的吗？或者有BLE开发经验？愿意参与下一代的开发吗

我现在用的是佳明520，带有导航功能的佳明1030要4500+元，太太贵了。。。
我有c++开发经验，应该BLE并没有经验，但是感觉不是大问题。愿意。

FASTSHIFT, 2021-07-28:
> > 有会写App的吗？或者有BLE开发经验？愿意参与下一代的开发吗
> 
> 我现在用的是佳明520，带有导航功能的佳明1030要4500+元，太太贵了。。。
> 我有c++开发经验，BLE并没有经验，但是感觉不是大问题。愿意。

第二代有计划加入导航功能，而且经过论证是完全可行的！
您可以尝试学习nRF52832，资料和现成的开发板都很多。

arinport, 2021-07-28:
@FASTSHIFT 

导航功能我可感兴趣了。可否详细赐教？如果使用定制的App如高德地图需要手机支持。如果您用自己的算法/离线地图的话，那就真的是厉害了。。。

另外如果有声音就比较困难（毕竟户外的功率要求挺高的），不知道您是计划用蓝牙耳机还是直接公放。

屏幕您这边第二代有任何计划吗？是否会使用AMOLED屏幕？我有测试过AMOLED屏幕直接放到车内可以投影到挡风玻璃上（1000nit真不是盖的）

FASTSHIFT, 2021-07-28:
> @FASTSHIFT
> 
> 导航功能我可感兴趣了。可否详细赐教？如果使用定制的App如高德地图需要手机支持。如果您用自己的算法/离线地图的话，那就真的是厉害了。。。
> 
> 另外如果有声音就比较困难（毕竟户外的功率要求挺高的），不知道您是计划用蓝牙耳机还是直接公放。
> 
> 屏幕您这边第二代有任何计划吗？是否会使用AMOLED屏幕？我有测试过AMOLED屏幕直接放到车内可以投影到挡风玻璃上（1000nit真不是盖的）

导航在手机上计算。
声音我觉得不是刚需，振动提示足够。但是加TTS也是可以的。
AMOLED分辨率多少？超过480*320不考虑，MCU带不动。

arinport, 2021-07-28:
@FASTSHIFT 

如果是在手机上计算您这边是打算定制高德地图吗？

FASTSHIFT, 2021-07-28:
> @FASTSHIFT
> 
> 如果是在手机上计算您这边是打算定制高德地图吗？

高德地图是待选之一

dankes, 2021-07-29:
参加开发

chenghongyao, 2021-07-30:
期待第二代，正学习工程，希望参与

HanfG, 2021-11-04:
希望可以参与第二代的开发！！！！
掌握java和python语言，开发过简单的安卓应用，但是UI太拉了；硬件懂一点点（软件工程自学硬件，基础功比较拉跨），AD熟练使用。

ZhongYic00, 2022-05-21:
> > 是不是可以建议采用e-ink屏幕以保障续航及高亮下清晰？
> 
> 墨水屏刷新体验不好，而且不适合在阳光下暴晒。个人比较喜欢追求体验，续航可以通过增加电池容量解决。

墨水屏对比度很高，应该很适合阳光下使用吧。看了一下可能的问题是暴晒导致泛黄，但只是部分屏幕的材料问题
考虑到刷新慢可以把动画去掉，然后通过局部刷新能够大大降低对MCU性能的要求

FChong96, 2022-08-22:
NXP的RT1060感觉性能满足不知道功耗如何

1061875478, 2022-08-28:
第二代开发还在进行吗

l-jesse, 2022-09-01:
> @FASTSHIFT
> 
> 导航功能我可感兴趣了。可否详细赐教？如果使用定制的App如高德地图需要手机支持。如果您用自己的算法/离线地图的话，那就真的是厉害了。。。
> 
> 另外如果有声音就比较困难（毕竟户外的功率要求挺高的），不知道您是计划用蓝牙耳机还是直接公放。
> 
> 屏幕您这边第二代有任何计划吗？是否会使用AMOLED屏幕？我有测试过AMOLED屏幕直接放到车内可以投影到挡风玻璃上（1000nit真不是盖的）

AMOLED 在哪里能买到？我在某宝搜了下有一家卖2.8寸，240*400带触摸，但是问客服说已经停产了。
还有 AMOLED 是不是都可以通过编程来调低刷新率到1HZ？这样的话可以放在省电模式对续航应该有帮助

vitrumzhu, 2024-01-12:
可以看看这个项目https://github.com/hishizuka/pizero_bikecomputer/

AlexZhang087, 2024-02-26:
各位大神们，这都24年，二代研发好了么