FASTSHIFT/X-TRACK #7: BOM  配单问题

Leo204-LKY, 2021-07-19:
什么也不懂，没有经验，想拿这个项目练练手（是真的什么都不懂QwQ）    
拿去淘宝 BOM 配单的时候店家说这些不够详细  
大佬可以帮忙补充下吗，谢谢！  
  
FootPrint | Comment | LibRef | Designator | Quantity | 问题
-- | -- | -- | -- | -- | --
Pad_B | BAT_IN | Pad | BAT+ | 1 | 型号？
Pad_B | GND | Pad | BAT- | 1 | 型号？
蜂鸣器 | Buzzer | BZ | BZ1 | 1 | 体积  多少V  有源无源
_SMD5020 |   |   |   |   | 什么感值
电容_C0805 | 4.7UF | cap | C1,   C3 | 2 |  
电容_C0603 | 104 | cap | C2,   C4, C7, C8, C13, C18, C19, C20, | 12 |  
电容_C0603 | 106 | cap | C5 | 1 |  
电容_C1206 | 22uF | cap | C6 | 1 |  
电容_C0603 | 105 | cap | C9,   C10, C11, C12, C16, C17 | 6 |  
电容_C0603 | 22pf | cap | C14,   C15 | 2 |  
LED_0603 | red | LED | D1 | 1 |  
二极管_SOD-323 | 1N4148 | Diode | D2,   D3, D4, D5 | 4 |  
二极管_SOD-323 | 1N4007 | Diode | D6 | 1 |  
开关按键 | SKRP | SW-PB | K1 | 1 | 体积 直插还是贴片
HDR1X4H | Header   4 | Header   4 | P2 | 1 | 排针还是排母
接插件 | Micro_SD | Micro_SD | P3 | 1 | 发图确认
接插件 | MICRO   USB | MICRO   USB | P4 | 1 | 发图确认
MOS_SOT-23 | AO3401 | MOSFET-P | Q1 | 1 |  
MOS_SOT-23 | AON3400 | MOSFET-N | Q2 | 1 |  
MOS_SOT-23 | AO3400 | MOSFET-N | Q3,   Q4 | 2 |  
电阻_R0603 | 2K | Res | R1 | 1 |  
电阻_R0603 | 1K | Res | R2 | 1 |  
电阻_R0603 | 20K | Res | R3,   R4, R5, R6 | 4 |  
电阻_R0603 | 10K | Res | R7,   R8, R11, R13, R14, R15, R16 | 7 |  
电阻_R0603 | 100R | Res | R9,   R10 | 2 |  
电阻_R0603 | 10R | Res | R12 | 1 |  
SOT23-5 | MCP73831 | MCP73831 | U1 | 1 |  
SOT23-5 | LP5907-3.3 | LP5907 | U2 | 1 |  
AT主控 | AT32F403ACGU | AT32F403CGU6 | U3 | 1 |  
串口 | CH340E | CH340E | U4 | 1 |  
GPS-HTZN | HT1818Z3G5L | HT1818Z3G5L | U5 | 1 |  
LGA- | LIS3MDL | LIS3DMDL | U6 | 1 |  
LGA- | LSM6DSM | LSM6DSM | U7 | 1 |  
ZJY154T- | IPS154-12 | IPS154-12 | U8 | 1 | 完整型号
SIQ-02FVS3 | Rotary   Encoder | Rotary   Encoder | U9 | 1 |  
晶振_3215 | 32.768K | XTAL_A | Y1 | 1 |  
Total |  |  |  | 67 |

FASTSHIFT, 2021-07-20:
只能帮到你这里
https://github.com/FASTSHIFT/X-TRACK/blob/main/Hardware/README.md

Leo204-LKY, 2021-07-20:
十分感谢！

Leo204-LKY, 2021-07-22:
emmm碰到新问题了
AT32F403ACGU7这个芯片没货，有可以代替的吗

FASTSHIFT, 2021-07-22:
@Leo204-LKY
可以换LQFP封装的AT32F403A，但是得会修改板子。

Leo204-LKY, 2021-07-22:
> @Leo204-LKY
> 可以换LQFP封装的AT32F403A，但是得会修改板子。

AT32F403AVCT7可以吗？

FASTSHIFT, 2021-07-22:
不行，FLASH容量至少512K，AT32F403AVCT7的FLASH大小是256KB。
另外，尽选小封装的MCU。

Leo204-LKY, 2021-07-22:
> 不行，FLASH容量至少512K，AT32F403AVCT7的FLASH大小是256KB。
> 另外，尽选小封装的MCU。

[image]

在tb上找到一张这样的表，麻烦大佬看一下，哪些符合要求
改板子的事我可以找人
实在抱歉，问题比较多

FASTSHIFT, 2021-07-22:
[image]
这个范围，体积要尽量小

Leo204-LKY, 2021-07-22:
> [image]
> 这个范围，体积要尽量小

好的，我去问问

zhihancn, 2021-07-24:
大佬有计算下来这些成本大致要多少钱呢？

FASTSHIFT, 2021-07-24:
@zhihancn 
大概不到300

Leo204-LKY, 2021-07-24:
@zhihancn 嗯，差不多
但目前找不到 AT32F403A 芯片，建议要做的话先备好芯片

zhihancn, 2021-07-25:
@FASTSHIFT
@Leo204-LKY  
哦哦哦，ok，谢谢大佬

zxuewu, 2021-08-05:
请问电路板是需要自己焊接吗？（本人不懂电路设计。）如果是，能把你上面提到的完善好的BOM配单发出来吗，我也想搞一下。谢谢！

Leo204-LKY, 2021-08-06:
> 请问电路板是需要自己焊接吗？（本人不懂电路设计。）如果是，能把你上面提到的完善好的BOM配单发出来吗，我也想搞一下。谢谢！

要自己焊接，不过tb也有加工的，不过价格就比较贵了
我最近没有时间做这个项目，所以只能暂时搁置了:/

zinwalin, 2021-08-12:
> @zhihancn
> 大概不到300

现在应该是不太够了

closed 2021-09-07 by FASTSHIFT