FASTSHIFT/X-TRACK #12: keil5 编译源码错误，请教

polyGithub, 2021-07-26:
> 

..\USER\App\Utils\ArduinoJson\src\ArduinoJson/Numbers/parseNumber.hpp(76): error:  #304: no instance of overloaded function "ArduinoJson6180_51::VariantData::setInteger" matches the argument list

..\Libraries\SdFat\src\SdFat.h(333): error:  #441: argument list for class template "SdFileSystem" is missing

polyGithub, 2021-07-26:
*** Using Compiler 'V5.05 (build 41)', folder: 'C:\Keil_v5\ARM\ARMCC\Bin'
*** Warning: Using Microlib and C++ file(s) might cause the linker to fail !
  
Rebuild target 'X-Track'
compiling DP_GPS.cpp...
compiling DP_Clock.cpp...
compiling DataProc.cpp...
compiling DP_SportStatus.cpp...
compiling App.cpp...
compiling DP_Power.cpp...
compiling DP_IMU.cpp...
compiling DP_MusicPlayer.cpp...
compiling DP_MAG.cpp...
compiling DP_Storage.cpp...
compiling DP_SysConfig.cpp...
compiling DP_TzConv.cpp...
compiling main.cpp...
compiling DP_Recorder.cpp...
..\Libraries\SdFat\src\SdFat.h(333): error:  #441: argument list for class template "SdFileSystem" is missing
             SdFileSystem::begin();
..\USER\App\Common\DataProc\DP_Recorder.cpp: 0 warnings, 1 error
compiling StartUpModel.cpp...
compiling StartUpView.cpp...
compiling DP_TrackFilter.cpp...
compiling StartUp.cpp...
compiling AppFactory.cpp...
compiling StatusBar.cpp...
compiling Dialplate.cpp...
compiling DialplateModel.cpp...
compiling DialplateView.cpp...
compiling SystemInfosModel.cpp...
compiling LiveMapModel.cpp...
compiling LiveMap.cpp...
compiling LiveMapView.cpp...
compiling TemplateModel.cpp...
compiling Template.cpp...
compiling SystemInfosView.cpp...
compiling font_bahnschrift_13.c...
compiling ResourcePool.cpp...
compiling TemplateView.cpp...
compiling font_bahnschrift_17.c...
compiling font_bahnschrift_32.c...
compiling font_bahnschrift_65.c...
compiling SystemInfos.cpp...
..\Libraries\SdFat\src\SdFat.h(333): error:  #441: argument list for class template "SdFileSystem" is missing
             SdFileSystem::begin();
..\USER\App\Pages\SystemInfos\SystemInfos.cpp: 0 warnings, 1 error
compiling font_agencyb_36.c...
compiling img_src_alarm.c...
compiling img_src_gps_arrow.c...
compiling img_src_gps_pin.c...
compiling img_src_battery.c...
compiling img_src_locate.c...
compiling img_src_menu.c...
compiling img_src_satellite.c...
compiling img_src_origin_point.c...
compiling img_src_stop.c...
compiling img_src_trip.c...
compiling img_src_pause.c...
compiling img_src_battery_info.c...
compiling img_src_bicycle.c...
compiling img_src_start.c...
compiling img_src_compass.c...
compiling img_src_gyroscope.c...
compiling img_src_map_location.c...
compiling img_src_system_info.c...
compiling img_src_storage.c...
compiling img_src_time_info.c...
compiling img_src_sd_card.c...
compiling lv_label_anim_effect.cpp...
[... 310 lines cut ...]
compiling delay.c...
compiling extend_SRAM.c...
compiling rtc.c...
compiling gpio.c...
compiling exti.c...
compiling timer.c...
compiling pwm.c...
compiling at32f4xx_adc.c...
compiling at32f4xx_bkp.c...
compiling at32f4xx_exti.c...
compiling at32f4xx_pwr.c...
compiling at32f4xx_rtc.c...
compiling at32f4xx_dma.c...
compiling at32f4xx_gpio.c...
compiling at32f4xx_rcc.c...
compiling at32f4xx_flash.c...
compiling at32f4xx_spi.c...
assembling startup_at32f403acgu7.s...
compiling misc.c...
compiling at32f4xx_usart.c...
compiling system_at32f4xx.c...
compiling at32f4xx_tim.c...
".\Objects\X-Track.axf" - 19 Error(s), 0 Warning(s).
Target not created.
Build Time Elapsed:  00:01:13

FASTSHIFT, 2021-07-26:
@polyGithub 
您的Keil的版本是？
务必使用Keil v5.25版本以上进行编译，旧编译器不一定能识别较新的C++语法

polyGithub, 2021-07-26:
> @polyGithub
> 您的Keil的版本是？
> 务必使用Keil v5.25版本以上进行编译，旧编译器不一定能识别较新的C++语法

好的，我的版本刚好是5.24的  我安装一个新的版本试试 ths

polyGithub, 2021-07-27:
使用keil 5.27版本 编译无问题

closed 2021-07-27 by polyGithub