FASTSHIFT/X-TRACK #17: Keil 编译报错

stig3, 2021-08-04:
环境 win10 keil 5.30 
AT32 DFP已经安装
然后报缺少库？

Error instantiating RTE components
Error #540: 'Keil::Device:Startup:1.0.1' component is not available for target 'X-Track'
Error #540: 'Keil::Device:StdPeriph Drivers:ADC:1.0.1' component is not available for target 'X-Track'
Error #540: 'Keil::Device:StdPeriph Drivers:BKPR:1.0.1' component is not available for target 'X-Track'
Error #540: 'Keil::Device:StdPeriph Drivers:DMA:1.0.1' component is not available for target 'X-Track'
Error #540: 'Keil::Device:StdPeriph Drivers:EXTI:1.0.1' component is not available for target 'X-Track'
Error #540: 'Keil::Device:StdPeriph Drivers:FMC:1.0.1' component is not available for target 'X-Track'
Error #540: 'Keil::Device:StdPeriph Drivers:Framework:1.0.1' component is not available for target 'X-Track'
Error #540: 'Keil::Device:StdPeriph Drivers:GPIO:1.0.1' component is not available for target 'X-Track'
Error #540: 'Keil::Device:StdPeriph Drivers:PWR:1.0.1' component is not available for target 'X-Track'
Error #540: 'Keil::Device:StdPeriph Drivers:RCC:1.0.1' component is not available for target 'X-Track'
Error #540: 'Keil::Device:StdPeriph Drivers:RTC:1.0.1' component is not available for target 'X-Track'
Error #540: 'Keil::Device:StdPeriph Drivers:SPI:1.0.1' component is not available for target 'X-Track'
Error #540: 'Keil::Device:StdPeriph Drivers:TIMER:1.0.1' component is not available for target 'X-Track'
Error #540: 'Keil::Device:StdPeriph Drivers:USART:1.0.1' component is not available for target 'X-Track'
Error #540: 'Keil::Device:System_AT32Fxx:1.0.1' component is not available for target 'X-Track'

ArtisticZhao, 2021-08-04:
Keil 版本多少？ 还有支持包能截个图么？

stig3, 2021-08-04:
> Keil 版本多少？ 还有支持包能截个图么？

Keil是5.30的
支持包截图如下
[image]

FASTSHIFT, 2021-08-04:
@stig3 使用这版本的pack
http://www.arterytek.com/download/Pack_Keil_AT32F4xx_CH_V1.3.4.zip

stig3, 2021-08-04:
> @stig3 请使用这版本的pack，新版本的标准库还没有验证过。
> http://www.arterytek.com/download/Pack_Keil_AT32F4xx_CH_V1.3.4.zip

不是这个问题，我之前就是安装的1.3.4，是报一样的问题，然后我去官网下载了1.3.7。还是一样的问题。

[image]

主要是这里报错
是不是还需要其他的软件包？能不能截一下你们的软件包，我逐个安装排查一下。

FASTSHIFT, 2021-08-04:
@stig3 
我是keil v5.25 + v1.3.4标准库

ArtisticZhao, 2021-08-04:
@stig3 我之前见过keil 5.30 编译失败的  要不你换成 keil 5.25 试试。 我5.25编译没有问题

LiangtaoZhong, 2021-08-05:
我之前也遇到了，换个keil版本重装一次。目前5.35没问题

stig3, 2021-08-05:
> 我之前也遇到了，换个keil版本重装一次。目前5.35没问题

谢谢，重找了一台Keil 5.25的电脑编译没问题
就是还不知道具体原因是什么

OuyangIPC, 2021-08-05:
是设置的问题吧，编译器提示某个部件对这个目标不可用

closed 2021-09-07 by FASTSHIFT