FASTSHIFT/X-TRACK #19: 关于充电动画问题

gyy0814, 2021-08-04:
大神您好 我现在用您提供的PCB以及V1.5的程序 制作了这个码表 使用起来很不错 但是 我发现我在充电的时候 屏幕并没有显示充电 在菜单栏里面依旧显示放电 但是电池电压会逐渐升高 充电指示灯也会点亮 但是没有充电动画 这个不知道是我焊接的问题 还是说程序里面没有这个充电检测  大概测量了一下 这个BAT_CHG_DET脚不充电时电压为0.0V 充电时大概为0.3V左右 大概问了一下 好像别人做出了也没有充电动画或者显示 您看看这个大概是什么问题 :D

FASTSHIFT, 2021-08-04:
是有充电动画的，在充电时BAT_CHG_DET(PA11)脚一定是高电平，量出来的0.3v不是正常电压

gyy0814, 2021-08-04:
啊这

gyy0814, 2021-08-04:
不过如果充电时BAT_CHG_DET为高电平 LED就亮不了啊 led是负级连接的CHG_DET

FASTSHIFT, 2021-08-04:
好像也是，电路设计有点小问题。但是程序里确实靠读PA11脚的电平判断的，说明STAT不是完全接地。

gyy0814, 2021-08-04:
[image]

FASTSHIFT, 2021-08-04:
嗯，正确的做法应该是测量USB_VCC，加两个分压电阻降到3.3V。

gyy0814, 2021-08-04:
而且我看您这个程序好像是靠程序里面电池的百分比来检测是否充电的  为啥不用这个充电芯片的状态脚来检测呢  我个人感觉这个程序里面的百分比在充电时是不太准的  例如我现在90左右的电量 我插入充电  百分比就变成了100 应该是充电时电池电压升高了 导致测出来的不准 此时断开充电器 电量会回落

FASTSHIFT, 2021-08-04:
精确的电量检测需要库仑计，感觉没太大必要为了这个增加成本。

gyy0814, 2021-08-04:
我的意思是 充电动画就不判断电量了 只判断U1的STAT和USB_VCC两个脚 :D

gyy0814, 2021-08-04:
个人认为写成这样更好 然后把IO配置为上拉

gyy0814, 2021-08-04:
[image]
[image]

gyy0814, 2021-08-04:
测试了一下 感觉挺好的 :D

gyy0814, 2021-08-04:
把PA11配置成内部上拉  然后检测IO 为低电平即为充电 高电平即为充满

FASTSHIFT, 2021-08-04:
可以

gyy0814, 2021-08-04:
:D

closed 2021-08-04 by gyy0814

gyy0814, 2021-08-04:
不知道大佬有没有准备把接口换成type-c 我尝试画了一个 但是本人不太会AD 所以说用立创EDA画了一个 布线乱七八糟的  希望大佬下次更新PCB的时候能考虑换成type-c

gyy0814, 2021-08-04:
[image]
[image]

FASTSHIFT, 2021-08-04:
下一代换Type-C，把调试脚也集成进去，厚度还能减一点。

gyy0814, 2021-08-04:
我把那个烧录排针换成四个贴片了啊哈哈  其实倒是想塞个st-link进去  不过 成本貌似....

gyy0814, 2021-08-04:
送去打板了  但愿不翻车

stig3, 2021-11-18:
看了一下，最新的程序充电检测那边还是没有改成pull-up
导致一直显示正在充电的状态
改成pull-up后充电状态显示正常
建议更新一下程序。

FASTSHIFT, 2021-11-18:
> 看了一下，最新的程序充电检测那边还是没有改成pull-up 导致一直显示正在充电的状态 改成pull-up后充电状态显示正常 建议更新一下程序。

已修复