FASTSHIFT/X-TRACK #43: 请问缓存一个页面是在哪里实现的？

priceee, 2021-12-12:
大佬你好，我在学习你的PageManager代码，我看到当SwitchTo一个页面时，会判断这个页面的isCached是否为真，若为真，则更改状态为WILL_APPEAR，这时就会设置该页面可见并启动页面切换动画。而若isCached为假，则会进入LOAD状态，这时会创建页面的root并执行onViewLoad()；
我想问一下此处的Cached是指什么？我为什么没有找到具体的执行某个函数将某个页面Cach的代码，只是看到有些地方将isCached赋值为真，有些地方将isCached赋值为假。
不知道我能不能说明白我的意思，希望大佬能解答我的疑惑。

FASTSHIFT, 2021-12-12:
cache的意思是当此页面退出时，不删除页面，而且将页面隐藏。当页面重新进入时，页面可以还原之前退出时的状态。

cache功能是可配置的：
* `SetCustomCacheEnable()` 设置自定义cache使能 (默认关闭)，用户手动管理页面的生命周期，页面不会被调度器自动删除。
* `SetCustomAutoCacheEnable()` 设置自动cache使能 (默认打开)，由调度器自动管理页面的声明周期：当push页面时，上一个页面的cache被打开；当页面pop时，被退出页面cache失能，页面被彻底删除。

注意，这两个函数推荐在`onCustomAttrConfig()`里进行调用，cache有关配置的请求在这里被同步到调度器中，**不要直接访问cache相关标志位**，可能会导致内存泄漏。

priceee, 2021-12-12:
谢谢大佬!