FASTSHIFT/X-TRACK #48: 请问如果需要更换软件I2C的GPIO口需要如何修改才行？

bbear2, 2021-12-20:
你好，我这边在查看代码后发现代码中的I2C是使用的软件模拟的，我在自己的设计中由于一些其他原因需要更换I2C的IO口。
具体是从PB6/PB7更换为PB8/PB9。
我的具体做法是修改mcu_config.h中的WIRE_SDA_PIN/WIRE_SCL_PIN的内容。但是我发现修改后无法正常产生I2C波形。
期间我尝试过打开WIRE_USE_FULL_SPEED_I2C宏，这样虽然有波形了，但是用逻辑分析仪抓取波形后发现波形不对劲，另外也和我的外设无法通讯。另外我也尝试在关闭WIRE_USE_FULL_SPEED_I2C的情况下修改IO口的输出类型为OUTPUT(原始代码是OUTPUT_OPEN_DRAIN)，这样修改后也能产生波形，但是依旧通讯失败。
同一个外设，我在使用PB6/PB7时是可以正常通讯的。

因此这里想请教下是不是我有漏了什么需要修改的地方？
期待回复~

FASTSHIFT, 2021-12-21:
检查是否为PB8/PB9添加了上拉电阻(3.3K~10K)，或者发生了短路。
在主频较高时，不建议打开`WIRE_USE_FULL_SPEED_I2C`，器件不一定能追上。I2C必须使用`OUTPUT_OPEN_DRAIN`开漏输出，不要使用推挽输出。

bbear2, 2021-12-21:
> 检查是否为PB8/PB9添加了上拉电阻(3.3K~10K)，或者发生了短路。 在主频较高时，不建议打开`WIRE_USE_FULL_SPEED_I2C`，器件不一定能追上。I2C必须使用`OUTPUT_OPEN_DRAIN`开漏输出，不要使用推挽输出。

我有为这两个引脚添加4.7k的上拉电阻，另外可以确认两个引脚没有短路。

FASTSHIFT, 2021-12-21:
使用以下代码检查是否可以正常翻转IO(两边是一个相反的方波)

```c
pinMode(PB8, OUTPUT_OPEN_DRAIN);
pinMode(PB9, OUTPUT_OPEN_DRAIN);

digitalWirte(PB8, LOW);
digitalWirte(PB9, HIGH);

while(1)
{
    togglePin(PB8);
    togglePin(PB9);
    delay(100);
}
```

FASTSHIFT, 2021-12-21:
我这边测试的更换PB8/PB9通信是正常的，请继续排查硬件问题。

bbear2, 2021-12-21:
> 使用以下代码检查是否可以正常翻转IO(两边是一个相反的方波)
> 
> ```c
> pinMode(PB8, OUTPUT_OPEN_DRAIN);
> pinMode(PB9, OUTPUT_OPEN_DRAIN);
> 
> digitalWirte(PB8, LOW);
> digitalWirte(PB9, HIGH);
> 
> while(1)
> {
>     togglePin(PB8);
>     togglePin(PB9);
>     delay(100);
> }
> ```

我有用c代码测试过这个反向的方波，看波形是正常的。仔细回忆了下，我在单步调试的时候发现一个异常点，当io口电平被置高时，io口的电压好像是1.8v，好像应该是3.3v才对

bbear2, 2021-12-21:
> 我这边测试的更换PB8/PB9通信是正常的，请继续排查硬件问题。

非常感谢你帮忙实际测试，我仔细检查后大致确认是硬件问题，后面做好新板子了再进一步测试。

closed 2021-12-21 by bbear2