FASTSHIFT/X-TRACK #52: 显示屏功耗问题

jppang, 2022-01-03:
X-TRACK在外面使用一段时间之后，SD卡壳有点烫手。
做了以下的基本功耗量测。所有的量测都是在USB接口5V供电的前提之下：

|器件组合   | 电流|
|---|---|
|完整X-TRACK  |  210 mA |

完整X-TRACK是指：CH340E+MCP73831+LP5907+MCU+IMU+SD+ENCODER+GPS+显示屏。

拆解功耗量测结果如下：

|器件组合 |   电流|
|---|---|
|完整X-TRACK去掉GPS |   190 mA|
|完整X-TRACK去掉显示屏 |   60 mA|
|完整X-TRACK去掉显示屏、GPS | 40 mA|

也就是说显示屏的电流功耗为150 mA。
以3颗LED的标准电流值是60 mA来说应该是太大了。
上面的电流值是在背光控制电路原理图当中的R12=10欧姆的状况之下。
用手边有的电阻更换之后量测如下：

|器件组合 |  电流|
|---|---|
|完整X-TRACK@R12=10欧姆  | 210 mA|
|完整X-TRACK@R12=27欧姆 |  110 mA|

这样换算一下，如果显示屏的背光驱动电流要60 mA的话，再加上驱动IC的功耗需求，
那么R12的电阻值应该在20～25欧姆之间即可。
使用R12=27欧姆的X-TRACK在阳光下并无大碍，可用。
不知大家有没有遇到过相同的问题？

FASTSHIFT, 2022-01-03:
几点小建议：
* 1.不建议边充边用，因为会增加充电芯片负担。
* 2.功耗测量以电池输出端测更合适，个人测试正常使用电流100mA左右。
* 3.可以通过调节PWM来控制显示屏亮度，并不需要修改阻值

jppang, 2022-01-03:
感谢你的建议。
再补充一点，上面的量测值是在没有接电池的状况之下的。
电池输出端的电流量测我再来试看看。

jppang, 2022-01-04:
用万用表从电池端的量测结果如下：

|器件组合 |  电流 |
|---|---|
|完整X-TRACK@R12=10欧姆 |   大于200 mA |
|完整X-TRACK@R12=27欧姆 |  140 mA |

手边的万用表的电流档位最高到200 mA。
看起来数值远大于版主的。
我再来排查是否我的电源电路有问题。

closed 2022-01-09 by jppang