FASTSHIFT/X-TRACK #53: Generate MAP Software

AhmadPanogari, 2022-01-05:
Hello,

I am a user from outside your country.

I have difficulty downloading the software to generate MAP (https://gitee.com/CrimsonHu/java_map_download/)

are you willing to add the software to the Tools or another way to easily download it?

Thanks.

FASTSHIFT, 2022-01-05:
@AhmadPanogari 
You can try to download [WeChat](https://www.wechat.com/) and register an account, add me as a friend ID: **FASTSHIFT**, I can send the map downloader to you.

halise3, 2022-06-17:
I need help to use the script for my page of [jrs express](https://jrstracking.com/) which is hosted on WordPress.

idreamsi, 2023-01-12:
> @AhmadPanogari You can try to download [WeChat](https://www.wechat.com/) and register an account, add me as a friend ID: **FASTSHIFT**, I can send the map downloader to you.
Thanks,
Is there no other way? Unfortunately, WeChat verification and registration SMS code is not received in my country!

idreamsi, 2023-01-26:
For easier download, I put the latest version in the Telegram channel below:

`https://t.me/java_map_download`