FASTSHIFT/X-TRACK #56: 请问有使用开源工具的计划吗？

jppang, 2022-01-24:
如题，这个专案的开发有使用开源工具的可能性吗？
固件：Visual Studio Code + Platformio ?

FASTSHIFT, 2022-01-24:
暂时没有，目前主线还是以Keil为主，调试方便一点，ARMCC编译出来的代码运行速度也快一些。

jppang, 2022-01-24:
了解。
有点可惜了。
这么好的开源专案却只能使用AD+Keil。
硬体部分我尝试用开源工具KiCad重新画了原理图跟PCB，最近好不容易搞定了。
固件部分想尝试是否能用VS Code+Platformio的方式来复制，不过ARTERY好像在Platformio里面没有BSP这些支援，不知版主有没有建议的做法？在VS Code+Platformio平台上。

FASTSHIFT, 2022-01-24:
雅特力的pack中有标准库源码，直接将源码复制出来与一起参与编译。

jppang, 2022-01-24:
了解。
谢谢版主的建议。

wc7086, 2022-04-22:
> 如题，这个专案的开发有使用开源工具的可能性吗？ 固件：Visual Studio Code + Platformio ?
PlatformIO还是不够完善，我打算使用 GNU Arm Embedded Toolchain 、make 、OpenOCD 交叉编译，目前在尝试使用这几个工具。

jppang, 2022-04-23:
> > 如题，这个专案的开发有使用开源工具的可能性吗？ 固件：Visual Studio Code + Platformio ?
> 
> PlatformIO 还是不够完善，我打算使用 GNU Arm Embedded Toolchain 、make 、OpenOCD 交叉编译，目前在尝试使用这几个工具。
> 
> https://www.v2ex.com/t/771173

感谢建议。

FASTSHIFT, 2022-04-23:
更新了一个Linux平台的移植 https://github.com/FASTSHIFT/X-TRACK/commit/ef456bcb1c87f9dfcd0c3ab3b819614036129460 ，可以试试。

jppang, 2022-08-06:
尝试将X-Track源码使用VisualStudio Code和Platformio用Arduino平台移植到ESP32。

经过一番折腾，终于成功编译产生固件了，然而一启动，画面却呈现破碎的状态。

如下的照片所示：
[image]

现在还在查找是硬件设计还是固件参数设定问题。不知有能看出问题点的能给建议吗？先感谢了！

我的源码在这里：https://github.com/jppang/ESP32_X-Track.git

软件夹里面有firmware档案夹，里面有固件。我使用的是ESP32 WROOM 32E模组整合在自己设计的开发板内，照片内的显示屏是240x280分辨率的，一样是ST7789驱动IC。相同的硬件来试lvgl的一些范例看起来都不会有画面破碎的现象。
HAL_Display跟lv_port_disp里面的源码我是整个改写了，使用的是TFT_eSPI的库，用double buffer启用DMA。其实用double buffer或single buffer，启用或不启用DMA都没有影响启动画面破碎的问题，只是描绘快慢问题而已。

手边如果有ESP32开发板和显示模组的同好如果有兴趣的话可否能尝试一下是否有相同现象？管脚设定在/lib/HAL/HAL_Config.h里面。

FASTSHIFT, 2022-08-06:
> 尝试将X-Track源码使用VisualStudio Code和Platformio用Arduino平台移植到ESP32。
>
> 经过一番折腾，终于成功编译产生固件了，然而一启动，画面却呈现破碎的状态。
>
> 如下的照片所示： [image]
>
> 现在还在查找是硬件设计还是固件参数设定问题。不知有能看出问题点的能给建议吗？先感谢了！ [...]

试试用240*280分辨率跑一下

FASTSHIFT, 2022-08-06:
> > 尝试将X-Track源码使用VisualStudio Code和Platformio用Arduino平台移植到ESP32。
> > 经过一番折腾，终于成功编译产生固件了，然而一启动，画面却呈现破碎的状态。
> > 如下的照片所示： [image]
> > 现在还在查找是硬件设计还是固件参数设定问题。不知有能看出问题点的能给建议吗？先感谢了！
> > 我的源码在这里：https://github.com/jppang/ESP32_X-Track.git
> > 软件夹里面有firmware档案夹，里面有固件。我使用的是ESP32 WROOM 32E模组整合在自己设计的开发板内，照片内的显示屏是240x280分辨率的，一样是ST7789驱动IC。相同的硬件来试lvgl的一些范例看起来都不会有画面破碎的现象。 HAL_Display跟lv_port_disp里面的源码我是整个改写了，使用的是TFT_eSPI的库，用double buffer启用DMA。其实用double buffer或single buffer，启用或不启用DMA都没有影响启动画面破碎的问题，只是描绘快慢问题而已。
> > 手边如果有ESP32开发板和显示模组的同好如果有兴趣的话可否能尝试一下是否有相同现象？管脚设定在/lib/HAL/HAL_Config.h里面。
> 
> 试试用240*280分辨率跑一下，感觉应该还是像驱动问题。

jppang, 2022-08-06:
昨天晚上终于找到导致画面破碎的问题点了。

原来是用TFT_eSPI库改写HAL_Display与lv_port_disp这两个源码其中设定描绘位置的setAddrWindow这个方法的引数，
Adafruit的库是定义左上角与右下角坐标，TFT_eSPI库是定义左上角坐标与宽长。由于方法名称与引数数目一摸一样，就放过没有改到，导致了这个查找好久甚至还求救各位的窘境。

[image]

上面的照片是换成240X240屏的组态，不过画面破碎问题跟屏的分辨率无关，是在找到上面的问题点之前怀疑所有可能的原因之下将240X280的屏换成240X240的。

不过，上面的画面周围有一圈白线，个别组件也有位移的状况。这个在用雅特力单片机与V2.1固件的时候是没有看到的，上面ESP32是用V2.7版源码，APP夹里面我都没有更动到。看来还有课题待解决了。

感谢版主的建议，谢谢。

FASTSHIFT, 2022-08-07:
> 昨天晚上终于找到导致画面破碎的问题点了。
>
> 原来是用TFT_eSPI库改写HAL_Display与lv_port_disp这两个源码其中设定描绘位置的setAddrWindow这个方法的引数， Adafruit的库是定义左上角与右下角坐标，TFT_eSPI库是定义左上角坐标与宽长。由于方法名称与引数数目一摸一样，就放过没有改到，导致了这个查找好久甚至还求救各位的窘境。
>
> [image] [...]

白色边框和偏移是默认主题设置的pad和border导致的，关闭`lv_conf.h`中的`LV_USE_THEME_DEFAULT`即可。

jppang, 2022-08-07:
依版主的建议，修改lv_conf.h的设定后画面就正常了。

[image]

感谢版主的建议！

closed 2024-05-14 by FASTSHIFT