FASTSHIFT/X-TRACK #61: 消息订阅发布框架使用中遇到的问题咨询

bbear2, 2022-03-17:
你好，我这边想参考IMU数据的传输流程新建一个sensor数据的传输。但是在仿真时遇到问题。

添加步骤描述如下：
我在HAL层增加了对sensor的HAL层操作，同时在HAL层添加了相关的update函数周期性的向Datacenter提交数据。
然后是在DataProc中参考IMU的添加了DP_sensor.cpp文件，也在HAL_Def.h中添加了Datacenter中sensor数据类型的结构定义。还有在DP_LIST.inc中添加了sensor的相关代码。
最后是界面获取Datacenter的数据，我在Init函数中创建了一个account,然后添加了一个订阅者。最后在需要获取数据的地方通过Pull操作获取相关数据。

当我在visual studio中进行仿真时，每次都会在DataCenter.cpp中的Find函数处报错。具体如下图：
[image]

下面是我做过的调试工作：
1 通过单步调试发现，程序是在进入到App_Init();后的while循环中的lv_timer_handler();后崩溃的。具体是下面箭头标出的这行代码：
```
            next = _lv_ll_get_next(&LV_GC_ROOT(_lv_timer_ll), LV_GC_ROOT(_lv_timer_act));

            if(lv_timer_exec(LV_GC_ROOT(_lv_timer_act))) {//<<<<<<<<<<<<<<<---------------------
                /*If a timer was created or deleted then this or the next item might be corrupted*/
                if(timer_created || timer_deleted) {
```
2 考虑到代码是在datacenter相关的代码中发生异常的，我尝试寻找调用了Find函数的代码。
发现在我注释掉界面中的Pull函数后，该异常会消失。通过查看代码发现，Pull函数中也的确有
调用Find函数。于是我尝试恢复调用Pull函数并加上断点，但是程序运行过程中没有停在该断点处，
最后还是发生了异常。

基于以上情况，我想了解下是否是我在添加代码的过程中有什么纰漏导致了这个错误。
热切期盼你们的回复。

FASTSHIFT, 2022-03-18:
代码和发生错误的调用栈发一下

bbear2, 2022-03-18:
你好，相关代码我已经上传到了这个[仓库](https://github.com/bbear2/X-TRACK/tree/datacenter_issue_debug)里面，
使用的是datacenter_issue_debug分支。可以直接运行其中的仿真器代码复现问题。

另外关于异常时的调用栈，我这边截了图，供你们参考：
[image]

FASTSHIFT, 2022-03-18:
Model没初始化，访问了account野指针。
建议改成这样：
[image]

FASTSHIFT, 2022-03-18:
在Ninebot上班？

bbear2, 2022-03-18:
非常感谢答复，刚刚验证过了没有问题。
哈哈哈，我不是ninebot员工，是我自己基于ninebot做的一个改装。

closed 2022-03-18 by bbear2