FASTSHIFT/X-TRACK #62: PageManager::Install相关问题

pockrt-m, 2022-03-25:
大佬好，我在`App.cpp`代码段中发现一次性加载了所有的page`manager.Install`，实际应用中发现这种操作会占用较多堆栈（我使用的单片机可操作堆栈较少），此处能否修改当某一page即将加载时在对其Install，随后page消失后对其Uninstall，尽可能减少堆栈的占用。如果可以该怎么操作？能否提供一个思路？

FASTSHIFT, 2022-03-26:
Install/Uninstall支持动态安装和卸载

pockrt-m, 2022-03-27:
请问是否还能从其他地方对程序进行精简呢，感觉这一套页面生命周期和订阅发布框架对内存占用较高，之前单纯使用LVGL可以同时使用SD卡和httpCilent，但是将其中的HAL驱动移植到本项目中后httpCilent便会报错`Memory allocation failed`，请大佬指点

FASTSHIFT, 2022-03-28:
有多少内存？
框架本身其实不太吃内存，可以先尝试把page cache关了。

pockrt-m, 2022-03-28:
我用的是ESP32，使用的是稚晖君peak的代码进行修改的；
之前一直用的Arduino IDE，因为能跑通就没太关注内存占用；
现在所有用到的page全部都`SetCustomCacheEnable(false)`；
目前使用`ESP.getFreeHeap()`函数获取空闲堆，如果不启用SD卡，`http.get()`操作前空闲堆为`126920`，操作后为`81608`；
如果开启SD卡，`http.get()`操作前空闲堆为`99472`，操作后为`98336`；
实测如果`http.get()`操作前空闲堆低于`110000`是不能成功的，只有大于`125000`的时候才能够百分百成功。

FASTSHIFT, 2022-03-28:
内存池使用太极限了，当碎片一多就很容易申请失败，系统稳定性无法保证。对于速度不敏感的地方考虑用PSRAM。

FASTSHIFT, 2022-03-28:
一个PageBase占用的内存只比一个lv_obj_t多十几个字节，远远比不上文件系统和协议栈的heap开销。可以用sizeof量一下你的Page类的内存占用，如果觉得占用较多，考虑使用动态分配用户数据。但是还是建议优先解决文件系统和协议栈内存开销过大的问题。

pockrt-m, 2022-03-29:
感谢，这部分计划加入PSRAM来解决了；
还想请问`AccountSystem::xxx_Commit(&xxx_info)`这一部分是需要在指定时间内刷新嘛，如果未在指定时间内刷新会出问题？若不刷新能否维持上一次的结果？

FASTSHIFT, 2022-03-29:
如果没有新的Commit提交，调用Pull会返回失败，需要在发起Pull那一端自己做个cache。

FASTSHIFT, 2022-04-28:
PSRAM只有SRAM十分之一的性能，不适合放draw buffer。如果追求帧率，考虑更换SRAM更大的MCU。

closed 2022-06-15 by FASTSHIFT