FASTSHIFT/X-TRACK #71: 中文加载问题

xiaoyi-hub, 2022-06-06:
您好，我用lvgl官方工具生成一些中文字体后用ResourceManager管理却无法显示中文？有解决方法吗

FASTSHIFT, 2022-06-06:
代码的文件编码用UTF8

xiaoyi-hub, 2022-06-06:
我用官方的工具生成的也是UTF8格式的，，字库直接定义在lv_config.h文件的宏是可以正常显示中文的[#define LV_FONT_CUSTOM_DECLARE LV_FONT_DECLARE(font_bahnschrift_18)],然而我把它添加至ResourceManager却无法显示，报的错误是 not found for U+4BB ，在生成的字库中所显示的字体对应的是U+4E00。是不是ResourceManager中Vector容器添加中文字模后再读出来会出了问题？(生成的字库文件除了中文其他数字和英文都是正常的)

FASTSHIFT, 2022-06-06:
ResourceManager其实就是一个简单的key-value匹配，是不是请求字体key的时候没有请求成功，返回了默认字体。

xiaoyi-hub, 2022-06-06:
请求成功了，字库中的英文和数字都显示正常，只有中文显示异常，楼主有空可以测试一下吗

xiaoyi-hub, 2022-06-07:
我可以确认生成的字库没有问题，你看如下图
[image]

[image]

[image]

我用自己生成的字库直接替换工程的font_bahnschrift_17，直接搞了一个demo中文显示是正常的

如下是运行x-track加载自己制作的中文字库运行结果
[image]

[image]

### 
昨天我做了很多测试
1.我用一个网友移植好的lvgl vs工程直接添加制作的中文字库也可以正常显示
2.把pageManager ResourceManager 单独抽出来移植到1的工程中，中文也显示正常
这问题太诡异，不知道为啥。我给你发附带的字库，楼主有空看一下
### 
/*******************************************************************************
 * Size: 18 px
 * Bpp: 4
 * Opts:
 ******************************************************************************/

#ifdef LV_LVGL_H_INCLUDE_SIMPLE
#include "lvgl.h"
#else
#include "lvgl/lvgl.h"
#endif

#ifndef FONT_BAHNSCHRIFT_18
#define FONT_BAHNSCHRIFT_18 1
#endif

#if FONT_BAHNSCHRIFT_18

 /*-----------------
  *    BITMAPS
  *----------------*/

  /*Store the image of the glyphs*/
static LV_ATTRIBUTE_LARGE_CONST const uint8_t glyph_bitmap[] = {
    /* U+0020 " " */

    /* U+0021 "!" */
    0x9f, 0x58, 0xf4, 0x8f, 0x37, 0xf2, 0x6f, 0x25,
    0xf1, 0x4f, 0x4, 0xf0, 0x3e, 0x2, 0xe0, 0x0,
    0x1, 0x31, 0xaf, 0x5a, 0xf5,

    /* U+0022 "\"" */
    0xbd, 0xe, 0xa9, 0xb0, 0xc8, 0x99, 0xc, 0x68,
    0x80, 0xa5,

    /* U+0023 "#" */
    0x0, 0x2f, 0x0, 0xb6, 0x0, 0x4, 0xd0, 0xd,
    0x40, 0x0, 0x6b, 0x0, 0xf2, 0x1, 0x18, 0xa1,
    0x2f, 0x20, 0xaf, 0xff, 0xff, 0xff, 0x73, 0x4c,
    0x94, 0x8e, 0x42, 0x0, 0xd4, 0x6, 0xb0, 0x1,
    0x2f, 0x32, 0xa9, 0x20, 0xaf, 0xff, 0xff, 0xff,
    0x73, 0x8e, 0x55, 0xe8, 0x52, 0x5, 0xc0, 0xe,
    0x30, 0x0, 0x7a, 0x0, 0xf1, 0x0, 0x9, 0x80,
    0x2f, 0x0, 0x0,

    /* U+0024 "$" */
    0x0, 0x1, 0x60, 0x0, 0x0, 0x2, 0xd0, 0x0,
    0x0, 0x6e, 0xfe, 0x40, 0x5, 0xfa, 0xec, 0xf1,
    0xb, 0xc2, 0xd2, 0xf6, 0xc, 0xc2, 0xd0, 0x84,
    0x8, 0xf6, 0xd0, 0x0, 0x1, 0xcf, 0xe1, 0x0,
[... 1378 lines cut ...]
/*-----------------
 *  PUBLIC FONT
 *----------------*/

 /*Initialize a public general font descriptor*/
 #if LV_VERSION_CHECK(8, 0, 0)
 const lv_font_t font_bahnschrift_17 = {
 #else
 lv_font_t font_bahnschrift_18 = {
 #endif
     .get_glyph_dsc = lv_font_get_glyph_dsc_fmt_txt,    /*Function pointer to get glyph's data*/
     .get_glyph_bitmap = lv_font_get_bitmap_fmt_txt,    /*Function pointer to get glyph's bitmap*/
     .line_height = 20,          /*The maximum line height required by the font*/
     .base_line = 3,             /*Baseline measured from the bottom of the line*/
 #if !(LVGL_VERSION_MAJOR == 6 && LVGL_VERSION_MINOR == 0)
     .subpx = LV_FONT_SUBPX_NONE,
 #endif
 #if LV_VERSION_CHECK(7, 4, 0) || LVGL_VERSION_MAJOR >= 8
     .underline_position = -2,
     .underline_thickness = 1,
 #endif
     .dsc = &font_dsc           /*The custom font data. Will be accessed by `get_glyph_bitmap/dsc` */
 };

 #endif /*#if FONT_BAHNSCHRIFT_18*/

FASTSHIFT, 2022-06-07:
文件编码不是指字库文件，而是存中文字符的**代码文件**，`DialplateView.cpp`的文件编码是`ANSI`，直接写中文字符肯定是对不上的，建议补充一下文件编码相关知识：https://zhuanlan.zhihu.com/p/343747149
[image]

改为`UTF-8`编码则可以正常显示：
[image]

[image]

你测试的中文可以显示是因为`LVGL.Simulator.cpp`是`UTF-8`编码的，和ResourceManager其实没有关系。
[image]

xiaoyi-hub, 2022-06-07:
受教了，非常感谢

closed 2022-06-15 by FASTSHIFT