FASTSHIFT/X-TRACK #72: 关于页面管理器 页面状态 （PageState） 切换的问题

zilongmix, 2022-06-10:
## 问题总结
在 `systeminfosView.cpp` 中存在函数 `onFocus` ,用于选项卡的切换。
[image]

我加入调试信息  `LV_LOG_USER("SYSTEMP-ONFOCUS\r\n");` 发现只有在每次键盘或鼠标输出时，才能进入到该函数中。而 `onFocus` 在如下函数中被调用。
[image]
查询代码最终在 `void PageManager::StateUpdate(PageBase* base)` 发现了页面状态切换的过程。
[image]

### 问题1：
[image]
在 `StateDidAppearExecute` 中返回的 **State** 为 `PAGE_STATE_ACTIVITY` ,按道理来说下次进入 `StateUpdate` 函数时，应该执行 `case PageBase::PAGE_STATE_ACTIVITY:` 分支，但调试发现，执行的一直是 `case PageBase::PAGE_STATE_DID_APPEAR:` 分支。
我想问问在哪里导致了状态的变化?

### 问题2：
[image]

为什么只有在鼠标键盘输出时，才会进入 `onViewDidAppear();` 函数，并打印我添加的信息。这个页面过程的具体代码在哪里？

FASTSHIFT, 2022-06-10:
`onFocus()`在第一次页面进入显示完成时被`onViewDidAppear()`调用一次，是为了让`group`的第一个`obj`进入`focus`状态。后面的`onFocus()`都是在拨动鼠标滚轮的时候被`group`调用。

zilongmix, 2022-06-10:
> `onFocus()`在第一次页面进入显示完成时被`onViewDidAppear()`调用一次，是为了让`group`的第一个`obj`进入`focus`状态并且移动到对应位置。后面的`onFocus()`都是在拨动鼠标滚轮的时候被`group`调用。
感谢！！

最终找到 `group` 回调函数如下：
[image]

**页面调度器理解**
`PageManager` 给用户提供的调度接口一共有三个
1. Pop
2. Push
3. BackHome
用户通过手动调用这三个接口，在不同页面之间切换。
页面 `Page` 的状态如下
```c
    /* Page state */
    typedef enum
    {
        PAGE_STATE_IDLE,
        PAGE_STATE_LOAD,
        PAGE_STATE_WILL_APPEAR,
        PAGE_STATE_DID_APPEAR,
        PAGE_STATE_ACTIVITY,
        PAGE_STATE_WILL_DISAPPEAR,
        PAGE_STATE_DID_DISAPPEAR,
        PAGE_STATE_UNLOAD,
        _PAGE_STATE_LAST
    } State_t;
```

不开启页面缓存下的切面调度流程如下图，所有页面都保存在一个 `页面栈` 中，栈顶为当前显示的页面。 
[image]
如果开启页面缓存，主要是少了 `PAGE_STATE_LOAD` 和 `PAGE_STATE_UNLOAD` 的两个状态，不用每次切换页面时，重新构造组件，但也增加了内存开销。
这样理解没错吧！

FASTSHIFT, 2022-06-10:
> 最终找到 `group` 回调函数如下： [image]
>
> **页面调度器理解** `PageManager` 给用户提供的调度接口一共有三个
>
> 1. Pop
> 2. Push
> 3. BackHome
>    用户通过手动调用这三个接口，在不同页面之间切换。
>    页面 `Page` 的状态如下
>
> ```c
>     /* Page state */
>     typedef enum [...]

是的，页面缓存的意义是为了保持页面上下文信息，切回的时候可以保持原样而不是被重置。

closed 2022-06-10 by zilongmix