FASTSHIFT/X-TRACK #73: About GPS Accuracy on Live MAP

AhmadPanogari, 2022-06-19:
I've successfully run this program with ESP32 and Ili9341 3.2 TFT LCD with Ublox-M8N GPS module.

However I have a problem with the result showing in Live Map (ARROW point), it doesn't look like what I see with the web based map even though it's the same longitude and latitude.

when viewed in Live View from Tile X and Y it does give the correct image but gives the wrong ARROW point.

is there any configuration to be able to adjust it?

may provide an offset configuration or other to match the actual situation.

Thanks very much.

FASTSHIFT, 2022-06-19:
@AhmadPanogari 
The program uses the [GCJ-02](https://en.wikipedia.org/wiki/Restrictions_on_geographic_data_in_China#GCJ-02) coordinate system by default, and you need to switch to the [WGS-84](https://en.wikipedia.org/wiki/World_Geodetic_System#WGS84)  coordinate system in your country.
Modification method:
Modify the `X-Track/USER/App/Config/Config.h` file in
`#define CONFIG_MAP_USE_WGS84_DEFAULT true`
or
Modify the `SystemSave.json` file in the SD card
`"sysConfig.mapWGS84": 1`

AhmadPanogari, 2022-07-02:
I've changed those settings and now it's accurate.

Thank you, sorry for the late response

closed 2022-07-02 by AhmadPanogari