FASTSHIFT/X-TRACK #76: 地图无法显示

dyufeimen, 2022-08-14:
按照教程的方法，模拟器里地图页面无法显示。请问是什么问题呢？
[image]

FASTSHIFT, 2022-08-14:
https://github.com/FASTSHIFT/X-TRACK/tree/main/Software#vs%E6%A8%A1%E6%8B%9F%E5%99%A8%E9%85%8D%E7%BD%AE

> PC模拟器上无法显示地图/显示颜色异常
* MAP地图文件夹放在X-TRACK工程的根目录`~/X-TRACK/MAP/`
* 修改**SystemSave.json**里的经纬度配置到希望显示地图的地方
* 确认生成的地图文件命名格式为`/z/x/y.png`
* 修改**SystemSave.json**的`"sysConfig.mapExtName"`项目，将`"bin"`改为`"png"`

dyufeimen, 2022-08-14:
> https://github.com/FASTSHIFT/X-TRACK/tree/main/Software#vs%E6%A8%A1%E6%8B%9F%E5%99%A8%E9%85%8D%E7%BD%AE
>
>
> * MAP地图文件夹放在X-TRACK工程的根目录`~/X-TRACK/MAP/`
> * 修改**SystemSave.json**里的经纬度配置到希望显示地图的地方
> * [...]

按照这几点修改了也不行，而且我通过调试发现在LiveMap.cpp中的MapTileContReload方法里给View.SetMapTileSrc(i, path);传入的path路径也是正确的，但是就是无法显示。
[image]

[image]

dyufeimen, 2022-08-14:
> 

我好像看了看path：/ovMAP/16/51705/x693w.png，但是我打开的路径是ovMAP/16/51705/26932/tile.png。难道是我转行地图图片的时候略过了那一步？

FASTSHIFT, 2022-08-14:
> > 
> 
> 我好像看了看path：/ovMAP/16/51705/x693w.png，但是我打开的路径是ovMAP/16/51705/26932/tile.png。难道是我转换地图图片的时候略过了那一步？

命名风格配错了，参考：
https://github.com/FASTSHIFT/X-TRACK/blob/main/Images/%E5%9C%B0%E5%9B%BE%E4%B8%8B%E8%BD%BD%E6%95%99%E7%A8%8B/4-%E5%9C%B0%E5%9B%BE%E4%B8%8B%E8%BD%BD%E9%85%8D%E7%BD%AE.png

dyufeimen, 2022-08-14:
果然是这个原因，感谢作者

closed 2022-10-05 by FASTSHIFT