FASTSHIFT/X-TRACK #78: 行者新出了个码表辰，我拆开看了一下它用的芯片，希望能给2代芯片选型一些参考

liyang5945, 2022-08-19:
600mAh电池号称能续航30h，应该是在不开启蓝牙和背光的情况下，即使这样它的工作电流也只有20mA左右，GPS+CPU+屏幕做到这个功耗，可以说是非常惊人了。我最近在尝试使用esp32-s3移植本项目，跑了下官方的蓝牙demo，工作电流都有100mA左右，然后买了个ATGM336H模块，发现它的工作电流也得50mA，再加上屏幕的话工作电流至少也要200mA了。辰刚发布我就参加了盲订活动，最近刚到货就把它拆开了，看看它用的是什么芯片

[image]

通过芯片上的编码查询之后，GPS芯片用的是这个: https://www.u-blox.com/en/product/ubx-m10050-chip 
CPU则是这个NRF52840: https://www.nordicsemi.com/products/nrf52840 Cortex™-M4 CPU 64MHz RAM只有256KB ，虽然性能有点低，但是只用它做个蓝牙芯片也是不错的。芯片价格我在淘宝也查了下，GPS那个只要二十多块，52840只要十几块钱，希望能给2代芯片选型一些参考。

greatws88, 2023-04-18:
esp32要用NIMBLE，传统蓝牙功耗太大了

FASTSHIFT, 2024-07-24:
二代码表使用 AT32F437 + NRF52832 方案。

closed 2024-07-24 by FASTSHIFT