FASTSHIFT/X-TRACK #80: Add external folder to visual studio

obitvn, 2022-10-22:
Sorry if this is not very relevant, but how can you add an external folder to visual studio, specifically here App and HAL folders.

obitvn, 2022-10-22:
i tried to edit the config file of vs, but it can only add files that don't create folder like you and i can't build
[image]
My folder looks like this
[image]
It doesn't show up like your folder, I tried searching for solution many times on google but to no avail
[image]

FASTSHIFT, 2022-10-22:
Use the filter view instead of the folder view.

obitvn, 2022-10-22:
thank you for the response, but it's not a view issue
[image]
i'm curious how did you add the external folder to the project? Add files one by one by dragging and dropping?

obitvn, 2022-10-22:
after adding the files, it seems that visual studio can't find the .h files due to the path structure
[image]
[image]
i guess this is because visual studio can't find this path in its root, i added it but vs still can't recognize it, how did you make visual studio work?
[image]

obitvn, 2022-10-24:
After failed experiment with x-track i decided to create a new VS project myself and add the files i needed to it, it worked, simply copy folder, ctrl+v into VS, then exclude Unnecessary files, add the path to that folder at Property>C/C++>Additional Include Directory
[image]

closed 2022-10-24 by obitvn