FASTSHIFT/X-TRACK #93: 编译的时候出现14个error

abcdeaaa123, 2023-07-31:
新手，用的5.38版本的keil，具体截图如下
[image]

代码完全没有修改，也没有动过配置文件，编译的时候报错大多数在sd卡附近，所有编译报错如下：

Build started: Project: proj
*** Using Compiler 'V5.05 update 1 (build 106)', folder: 'D:\Keil_v5\ARM\ARMCC\Bin'
Build target 'X-Track'
compiling DP_IMU.cpp...
..\USER\App\Common\DataProc\DP_IMU.cpp(6): error:  #2958: the body of a value-returning lambda with no explicit return type must be a single return statement
      HAL::IMU_SetCommitCallback([](void* info, void* userData){
..\USER\App\Common\DataProc\DP_IMU.cpp: 0 warnings, 1 error
compiling DP_MAG.cpp...
..\USER\App\Common\DataProc\DP_MAG.cpp(6): error:  #2958: the body of a value-returning lambda with no explicit return type must be a single return statement
      HAL::MAG_SetCommitCallback([](void* info, void* userData){
..\USER\App\Common\DataProc\DP_MAG.cpp: 0 warnings, 1 error
compiling LIS3MDL.cpp...
..\Libraries\LIS3MDL\LIS3MDL.cpp(55): error:  #19: extra text after expected end of number
              if (sa1 != sa1_low && testReg(LIS3MDL_SA1_HIGH_ADDRESS, WHO_AM_I) == LIS3MDL_WHO_ID)
..\Libraries\LIS3MDL\LIS3MDL.cpp(64): error:  #19: extra text after expected end of number
              else if (sa1 != sa1_high && testReg(LIS3MDL_SA1_LOW_ADDRESS, WHO_AM_I) == LIS3MDL_WHO_ID)
..\Libraries\LIS3MDL\LIS3MDL.cpp(86): error:  #19: extra text after expected end of number
          address = (sa1 == sa1_high) ? LIS3MDL_SA1_HIGH_ADDRESS : LIS3MDL_SA1_LOW_ADDRESS;
..\Libraries\LIS3MDL\LIS3MDL.cpp(86): error:  #19: extra text after expected end of number
          address = (sa1 == sa1_high) ? LIS3MDL_SA1_HIGH_ADDRESS : LIS3MDL_SA1_LOW_ADDRESS;
..\Libraries\LIS3MDL\LIS3MDL.cpp: 0 warnings, 4 errors
compiling rt_sys.cpp...
..\Libraries\SdFat\src\SdFat.h(333): error:  #441: argument list for class template "SdFileSystem" is missing
             SdFileSystem::begin();
..\USER\HAL\rt_sys.cpp: 0 warnings, 1 error
compiling HAL_SD_CARD.cpp...
..\Libraries\SdFat\src\SdFat.h(333): error:  #441: argument list for class template "SdFileSystem" is missing
             SdFileSystem::begin();
..\USER\HAL\HAL_SD_CARD.cpp: 0 warnings, 1 error
compiling HAL.cpp...
..\Libraries\SdFat\src\SdFat.h(333): error:  #441: argument list for class template "SdFileSystem" is missing
             SdFileSystem::begin();
..\USER\HAL\HAL.cpp: 0 warnings, 1 error
compiling lv_port_fs_sdfat.cpp...
..\Libraries\SdFat\src\SdFat.h(333): error:  #441: argument list for class template "SdFileSystem" is missing
             SdFileSystem::begin();
..\USER\lv_port\lv_port_fs_sdfat.cpp: 0 warnings, 1 error
compiling HAL_FaultHandle.cpp...
..\Libraries\SdFat\src\SdFat.h(333): error:  #441: argument list for class template "SdFileSystem" is missing
             SdFileSystem::begin();
..\USER\HAL\HAL_FaultHandle.cpp: 0 warnings, 1 error
compiling DP_Recorder.cpp...
..\Libraries\SdFat\src\SdFat.h(333): error:  #441: argument list for class template "SdFileSystem" is missing
             SdFileSystem::begin();
..\USER\App\Common\DataProc\DP_Recorder.cpp: 0 warnings, 1 error
compiling SystemInfos.cpp...
..\Libraries\SdFat\src\SdFat.h(333): error:  #441: argument list for class template "SdFileSystem" is missing
             SdFileSystem::begin();
..\USER\App\Pages\SystemInfos\SystemInfos.cpp: 0 warnings, 1 error
compiling StartupView.cpp...
..\Libraries\SdFat\src\SdFat.h(333): error:  #441: argument list for class template "SdFileSystem" is missing
             SdFileSystem::begin();
..\USER\App\Pages\Startup\StartupView.cpp: 0 warnings, 1 error
".\Objects\X-Track.axf" - 14 Error(s), 0 Warning(s).
Target not created.
Build Time Elapsed:  00:00:06

求大佬们能给一点指导，谢谢了！

abcdeaaa123, 2023-07-31:
补充一下尝试过的措施。。查了一下有说在配置里加上--cpp11参数的，加完变成700多error了。。。

FASTSHIFT, 2023-07-31:
编译环境问题，尝试用Keil v5.36试试。

closed 2023-07-31 by abcdeaaa123

w6c6x6, 2024-10-22:
我换了5.36的版本还是报错除了换版本还做了什么操作吗