FASTSHIFT/X-TRACK #97: 支持嵌入式linux平台么

justdomyself, 2023-09-22:
如题

FASTSHIFT, 2023-09-22:
支持的，见目录：https://github.com/FASTSHIFT/X-TRACK/tree/main/Software/X-Track/Linux，稍微改改makefile就可以用。

justdomyself, 2023-09-22:
> 支持的，见目录：[Software/X-Track/Linux](https://github.com/FASTSHIFT/X-TRACK/tree/main/Software/X-Track/Linux)，稍微改改makefile就可以用。

编译报错：
ome/alientek/Desktop/cmake/X-TRACK/Software/X-Track/Linux/../Simulator/LVGL.Simulator/lv_drivers/sdl/sdl_common.c:5:
/home/alientek/Desktop/cmake/X-TRACK/Software/X-Track/Linux/../Simulator/LVGL.Simulator/lv_drivers/sdl/sdl_common.h:33:10: error: #include expects "FILENAME" or <FILENAME>
   33 | #include SDL_INCLUDE_PATH
      |          ^~~~~~~~~~~~~~~~
/home/alientek/Desktop/cmake/X-TRACK/Software/X-Track/Linux/../Simulator/LVGL.Simulator/lv_drivers/sdl/sdl_common.h:86:34: error: unknown type name ‘SDL_Event’
   86 | int quit_filter(void * userdata, SDL_Event * event);
      |                                  ^~~~~~~~~
/home/alientek/Desktop/cmake/X-TRACK/Software/X-Track/Linux/../Simulator/LVGL.Simulator/lv_drivers/sdl/sdl_common.h:88:20: error: unknown type name ‘SDL_Event’
   88 | void mouse_handler(SDL_Event * event);
      |                    ^~~~~~~~~
/home/alientek/Desktop/cmake/X-TRACK/Software/X-Track/Linux/../Simulator/LVGL.Simulator/lv_drivers/sdl/sdl_common.h:89:25: error: unknown type name ‘SDL_Event’
   89 | void mousewheel_handler(SDL_Event * event);
      |                         ^~~~~~~~~

FASTSHIFT, 2023-09-22:
已更新，可以拉最新代码试试。

justdomyself, 2023-09-22:
> 已更新，可以拉最新代码试试。
gcc编译没有问题，交叉编译提示nullptr错误：
In file included from /home/alientek/Desktop/cmake/X-TRACK/Software/X-Track/Linux/../USER/App/Pages/AppFactory.h:23:0,
                 from /home/alientek/Desktop/cmake/X-TRACK/Software/X-Track/Linux/../USER/App/App.cpp:26:
/home/alientek/Desktop/cmake/X-TRACK/Software/X-Track/Linux/../USER/App/Utils/PageManager/PageManager.h: At global scope:
/home/alientek/Desktop/cmake/X-TRACK/Software/X-Track/Linux/../USER/App/Utils/PageManager/PageManager.h:105:40: error: 'nullptr' was not declared in this scope
     PageManager(PageFactory* factory = nullptr);
                                        ^
/home/alientek/Desktop/cmake/X-TRACK/Software/X-Track/Linux/../USER/App/Utils/PageManager/PageManager.h:115:69: error: 'nullptr' was not declared in this scope
     bool Replace(const char* name, const PageBase::Stash_t* stash = nullptr);
                                                                     ^
/home/alientek/Desktop/cmake/X-TRACK/Software/X-Track/Linux/../USER/App/Utils/PageManager/PageManager.h:116:66: error: 'nullptr' was not declared in this scope
     bool Push(const char* name, const PageBase::Stash_t* stash = nullptr);
                                                                  ^
/home/alientek/Desktop/cmake/X-TRACK/Software/X-Track/Linux/../USER/App/Utils/PageManager/PageManager.h:172:85: error: 'nullptr' was not declared in this scope
     bool SwitchTo(PageBase* base, bool isEnterAct, const PageBase::Stash_t* stash = nullptr);

FASTSHIFT, 2023-09-22:
make的时候指定一下编译器：
```sh
make CC=<C编译器名字> CXX=<C++编译器名字>
```

justdomyself, 2023-09-25:
> make的时候指定一下编译器：
> 
> ```shell
> make CC=<C编译器名字> CXX=<C++编译器名字>
> ```
> 
> 你的C++编译器得支持C++11标准。

开了的，现在报这个错：

CC /home/alientek/Desktop/cmake/X-TRACK/Software/X-Track/Linux/../USER/App/Utils/lv_ext/lv_anim_timeline_wrapper.c
In file included from /home/alientek/Desktop/cmake/X-TRACK/Software/X-Track/Linux/../Simulator/LVGL.Simulator/HAL/HAL_Power.cpp:1:
/home/alientek/Desktop/cmake/X-TRACK/Software/X-Track/Linux/../Simulator/LVGL.Simulator/HAL/HAL.h:26:10: fatal error: Common/HAL/HAL.h: No such file or directory
   26 | #include "Common/HAL/HAL.h"
      |          ^~~~~~~~~~~~~~~~~~
In file included from /home/alientek/Desktop/cmake/X-TRACK/Software/X-Track/Linux/../Simulator/LVGL.Simulator/HAL/HAL_Audio.cpp:1:
/home/alientek/Desktop/cmake/X-TRACK/Software/X-Track/Linux/../Simulator/LVGL.Simulator/HAL/HAL.h:26:10: fatal error: Common/HAL/HAL.h: No such file or directory
   26 | #include "Common/HAL/HAL.h"
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.

FASTSHIFT, 2023-09-25:
试了一下树莓派的工具链编译没有问题，你的编译命令是什么？
```sh
make CC=/opt/cross-pi-gcc/bin/arm-linux-gnueabihf-gcc CXX=/opt/cross-pi-gcc/bin/arm-linux-gnueabihf-g++ -j12
```
这个报错就是头文件引用路径找不到，可以自己先排查一下。

justdomyself, 2023-09-25:
> ```shell
> a
> ```

1、刷新环境变量
 source /opt/fsl-imx-xwayland/5.4-zeus/environment-setup-aarch64-poky-linux 

2、查看环境变量
**env | grep CC** 

QT_ACCESSIBILITY=1
OE_QMAKE_RCC=/opt/fsl-imx-xwayland/5.4-zeus/sysroots/x86_64-pokysdk-linux/usr/bin/rcc
CLANGCC=aarch64-poky-linux-clang  -mcpu=cortex-a53+crc+crypto -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security -mlittle-endian --sysroot=/opt/fsl-imx-xwayland/5.4-zeus/sysroots/aarch64-poky-linux
OE_QMAKE_CC=aarch64-poky-linux-gcc  -mcpu=cortex-a53+crc+crypto -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/opt/fsl-imx-xwayland/5.4-zeus/sysroots/aarch64-poky-linux
CC=aarch64-poky-linux-gcc  -mcpu=cortex-a53+crc+crypto -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/opt/fsl-imx-xwayland/5.4-zeus/sysroots/aarch64-poky-linux

**env | grep CXX**
OE_QMAKE_CXX=aarch64-poky-linux-g++  -mcpu=cortex-a53+crc+crypto -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/opt/fsl-imx-xwayland/5.4-zeus/sysroots/aarch64-poky-linux
CXX=aarch64-poky-linux-g++  -mcpu=cortex-a53+crc+crypto -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/opt/fsl-imx-xwayland/5.4-zeus/sysroots/aarch64-poky-linux
CXXFLAGS= -O2 -pipe -g -feliminate-unused-debug-types 
OE_QMAKE_CXXFLAGS= -O2 -pipe -g -feliminate-unused-debug-types 
CLANGCXX=aarch64-poky-linux-clang++  -mcpu=cortex-a53+crc+crypto -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security -mlittle-endian --sysroot=/opt/fsl-imx-xwayland/5.4-zeus/sysroots/aarch64-poky-linux

3、直接编译，输入make -j4

FASTSHIFT, 2023-09-25:
在makefile加点log看一下`INCLUDE_CFG`变量头文件引用路径有没有问题。

justdomyself, 2023-09-25:
> 在makefile加点log看一下`INCLUDE_CFG`变量头文件引用路径有没有问题。

这是打印的INCLUDE_CFG的值：
-I/home/alientek/Desktop/cmake/X-TRACK/Software/X-Track/Linux/../Simulator/LVGL.Simulator/ -I/home/alientek/Desktop/cmake/X-TRACK/Software/X-Track/Linux/../USER/App -I/home/alientek/Desktop/cmake/X-TRACK/Software/X-Track/Linux/../USER/App/Utils/ArduinoJson/src

FASTSHIFT, 2023-09-25:
引用路径没问题，那应该就是编译环境问题了，自己查一下吧。

justdomyself, 2023-09-26:
> 引用路径没问题，那应该就是编译环境问题了，自己查一下吧。

编译环境都是原厂提供的，编译其他的源码没出现过这种问题呢

closed 2024-07-24 by FASTSHIFT