FASTSHIFT/X-TRACK #103: 无法获取手势动作

Ccuso4, 2024-01-09:
大佬打扰了，我是在PC上运行稍微修改了下代码想适配触摸屏的手势动作，但是每次都只返回`LV_EVENT_PRESSED`，以下是修改的code。
```
void Dialplate::onViewLoad()
{
    Model.Init();
    View.Create(_root);
    AttachEvent(_root);
    AttachEvent(View.ui.topInfo.cont);
    //AttachEvent(View.ui.btnCont.btnRec);
    //AttachEvent(View.ui.btnCont.btnMenu);
}
......
......
void Dialplate::onEvent(lv_event_t* event)
{
    if (code == LV_EVENT_GESTURE)
    {
        printf("GESTURE %d", code);

        if (lv_indev_get_gesture_dir(lv_indev_get_act()) == LV_DIR_LEFT) {
            instance->_Manager->Push("Pages/SystemInfos");
        }
    }
    if (code == LV_EVENT_PRESSED)
    {
        printf("LV_EVENT_PRESSED %d \n", code);
    }
}
```
打印值：
```
[PM][Warn] Page PagePrev(Pages/Startup) is busy
LV_EVENT_PRESSED 1
LV_EVENT_PRESSED 1
LV_EVENT_PRESSED 1
LV_EVENT_PRESSED 1
LV_EVENT_PRESSED 1
LV_EVENT_PRESSED 1
```
感谢

FASTSHIFT, 2024-01-09:
@Ccuso4 
可以看一下lvgl的手势识别的代码实现，见`lvgl\src\core\lv_indev.c`中的`indev_gesture`函数，里面计算的得出的`gesture_obj`为什么不是想要的那个。

Ccuso4, 2024-01-10:
> @Ccuso4 可以看一下lvgl的手势识别的代码实现，见`lvgl\src\core\lv_indev.c`中的`indev_gesture`函数，里面计算的得出的`gesture_obj`为什么不是想要的那个。

感谢你的回复，我在`lvgl\src\core\lv_indev.c`里加了一些log，发现手势识别是没有问题的，只是传过去的对象不一致，不太明白为啥。以下是修改代码和log
```
void indev_gesture(_lv_indev_proc_t * proc)
{

    if(proc->types.pointer.scroll_obj) return;
    if(proc->types.pointer.gesture_sent) return;

    lv_obj_t * gesture_obj = proc->types.pointer.act_obj;

    /*If gesture parent is active check recursively the gesture attribute*/
    while(gesture_obj && lv_obj_has_flag(gesture_obj, LV_OBJ_FLAG_GESTURE_BUBBLE)) {
        gesture_obj = lv_obj_get_parent(gesture_obj);
    }
    if(gesture_obj == NULL) return;
    if((LV_ABS(proc->types.pointer.vect.x) < indev_act->driver->gesture_min_velocity) &&
       (LV_ABS(proc->types.pointer.vect.y) < indev_act->driver->gesture_min_velocity)) {
        proc->types.pointer.gesture_sum.x = 0;
        proc->types.pointer.gesture_sum.y = 0;
    }

    /*Count the movement by gesture*/
    proc->types.pointer.gesture_sum.x += proc->types.pointer.vect.x;
    proc->types.pointer.gesture_sum.y += proc->types.pointer.vect.y;

    if((LV_ABS(proc->types.pointer.gesture_sum.x) > indev_act->driver->gesture_limit) ||
       (LV_ABS(proc->types.pointer.gesture_sum.y) > indev_act->driver->gesture_limit)) {

        proc->types.pointer.gesture_sent = 1;

        if(LV_ABS(proc->types.pointer.gesture_sum.x) > LV_ABS(proc->types.pointer.gesture_sum.y)) {
            if (proc->types.pointer.gesture_sum.x > 0)
            {
                proc->types.pointer.gesture_dir = LV_DIR_RIGHT;
                LV_LOG_USER("LV_DIR_RIGHT !!!!!!");
            }
            else
            {
                proc->types.pointer.gesture_dir = LV_DIR_LEFT;
                LV_LOG_USER("LV_DIR_LEFT !!!!!!");
            }
        }
        else {
            if(proc->types.pointer.gesture_sum.y > 0)
            {
                proc->types.pointer.gesture_dir = LV_DIR_BOTTOM;
                LV_LOG_USER("LV_DIR_BOTTOM !!!!!!");
            }
            else
            {
                proc->types.pointer.gesture_dir = LV_DIR_TOP;
                LV_LOG_USER("LV_DIR_TOP !!!!!!");
            }
        }

        lv_event_send(gesture_obj, LV_EVENT_GESTURE, indev_act);
        LV_LOG_USER("gesture_obj:%d,  LV_EVENT_GESTURE:%d", gesture_obj, LV_EVENT_GESTURE);
        if(indev_reset_check(proc)) return;
    }
```
log
```
[User]  (1275390.365, +7538)     onEvent: obj:-1322611424 LV_EVENT_PRESSED:1   (in Dialplate.cpp line #249)
[User]  (1275390.458, +93)       indev_gesture: LV_DIR_LEFT !!!!!!      (in lv_indev.c line #1149)
[User]  (1275390.459, +1)        indev_gesture: gesture_obj:-1322644160,  LV_EVENT_GESTURE:12   (in lv_indev.c line #1166)
```
感谢

Ccuso4, 2024-01-10:
> 感谢你的回复，我在`lvgl\src\core\lv_indev.c`里加了一些log，发现手势识别是没有问题的，只是传过去的对象不一致，不太明白为啥。以下是修改代码和log
>
> ```
> void indev_gesture(_lv_indev_proc_t * proc)
> {
>
>     if(proc->types.pointer.scroll_obj) return; [...]

我发现将手势对象传递给父对象的flag禁掉后就可以了`lv_obj_clear_flag(cont, LV_OBJ_FLAG_GESTURE_BUBBLE);`

FASTSHIFT, 2024-01-10:
Good!

closed 2024-01-10 by Ccuso4