FASTSHIFT/X-TRACK #104: 需要提前将GPX文件准备好吗？那么如果第一次去某个地方，没有GPX文件，如何处理？

Doing-code, 2024-03-14:
代码路径：`X-Track\Simulator\LVGL.Simulator\HAL\HAL_GPS.cpp`

```c++
static bool Parser_Init(GPX_Parser* parser, FileInfo_t* info) {
...
    lv_fs_res_t res = lv_fs_open(&info->file, CONFIG_TRACK_VIRTUAL_GPX_FILE_PATH, LV_FS_MODE_RD);
...
}
```
如果是第一次骑行，如何准备GPX文件？如果去GPX没有覆盖的地方，程序还会正常记录速度、耗时等数据信息吗？

如果第一次没有GPX文件，运动记录操作跑不起来，这种情况如何解决？？一定要实现准备GPX吗？

Doing-code, 2024-03-14:
```txt
onRecord: GPS has not ready, can't start record
```
如果没有GPX文件，会出现如上错误，一定要事先准备.GPX，才能启动开始按钮吗 ？

FASTSHIFT, 2024-03-14:
> 代码路径：`X-Track\Simulator\LVGL.Simulator\HAL\HAL_GPS.cpp`
>
> ```c++
> static bool Parser_Init(GPX_Parser* parser, FileInfo_t* info) {
> ...
>     lv_fs_res_t res = lv_fs_open(&info->file, [...]

这个是PC模拟器的逻辑。因为PC上没有真实的GPS设备来产生坐标数据，所以通过回放GPX文件的方式来模拟GPS数据。
在真实设备上，GPS数据取自GPS模块，见代码文件[HAL_GPS.cpp](https://github.com/FASTSHIFT/X-TRACK/blob/main/Software/X-Track/USER/HAL/HAL_GPS.cpp)，不需要准备GPX文件。

FASTSHIFT, 2024-03-14:
> ```
> onRecord: GPS has not ready, can't start record
> ```
> 
> 如果没有GPX文件，会出现如上错误，一定要事先准备.GPX，才能启动开始按钮吗 ？

这个报错的含义是，要求GPS要先定位上（有坐标数据产生），才能开始进行轨迹记录，轨迹记录器不关心实际的GPS数据源。PC模拟器需要导入GPX文件才能产生GPS数据，在真实设备上是GPS连接成功就能产生GPS数据。

closed 2024-05-14 by FASTSHIFT