FASTSHIFT/X-TRACK #106: gpxsee查看历史轨迹的时候使用高德瓦片地图显示位置偏移

magic-zjk, 2024-05-14:
下载的高德瓦片地图在码表中显示正常。

导出历史轨迹通过gpxsee查看时，选择了自定义的瓦片服务器（使用该高德瓦片地图）显示位置发生偏移。但是使用gpxsee自带的Open Topo Map地图能正常显示。

不知道作者有没有遇到过这种情况？

FASTSHIFT, 2024-05-14:
地图偏移是因为gpxsee只支持标准的WGS84坐标系统，而国内的地图使用的是GCJ02坐标系统，是坐标系统不一致导致发生了偏移。
gpxsee的自带地图显示正常是因为使用了国外的地图源，国外的地图源都是WGS84坐标系统。
码表显示正确是因为做了坐标转换。

magic-zjk, 2024-05-17:
> 地图偏移是因为gpxsee只支持标准的WGS84坐标系统，而国内的地图使用的是GCJ02坐标系统，是坐标系统不一致导致发生了偏移。 gpxsee的自带地图显示正常是因为使用了国外的地图源，国外的地图源都是WGS84坐标系统。 码表显示正确是因为做了坐标转换。

@FASTSHIFT 
在没有梯子的情况下Open Topo Map地图勉强能显示。请教一下有没有其他办法使用自定义的瓦片服务器也能正常显示呢？

FASTSHIFT, 2024-05-17:
> > 地图偏移是因为gpxsee只支持标准的WGS84坐标系统，而国内的地图使用的是GCJ02坐标系统，是坐标系统不一致导致发生了偏移。 gpxsee的自带地图显示正常是因为使用了国外的地图源，国外的地图源都是WGS84坐标系统。 码表显示正确是因为做了坐标转换。
> 
> @FASTSHIFT 在没有梯子的情况下Open Topo Map地图勉强能显示。请教一下有没有其他办法使用自定义的瓦片服务器也能正常显示呢？

没了解过，可以在gpxsee上发个issue问问。

closed 2024-07-24 by FASTSHIFT

Reskip, 2024-08-19:
> 下载的高德瓦片地图在码表中显示正常。
> 
> 导出历史轨迹通过gpxsee查看时，选择了自定义的瓦片服务器（使用该高德瓦片地图）显示位置发生偏移。但是使用gpxsee自带的Open Topo Map地图能正常显示。
> 
> 不知道作者有没有遇到过这种情况？

导入前可以先进行一次坐标转换，开源的函数很多
网页端工具也有，可以参考 https://mp.weixin.qq.com/s/4jLedSmkjJNsDhx-zDxwWg