FASTSHIFT/X-TRACK #125: Unable to see map in ubuntu environment

Prajyot01, 2025-08-06:
Hello ,
I have run the X-Track linuxSDL2 but i am not able to see map. i have created Map folder in root directory and put there png files . Also the Systemsave.json and other files are not generated automatically. kindly help us to resolve this issue.

[image]

[image]

FASTSHIFT, 2025-08-12:
You can try modifying the default longitude and latitude in [Config.h](https://github.com/FASTSHIFT/X-TRACK/blob/main/Software/X-Track/USER/App/Config/Config.h#L44) and set them to the location where you want to display the map.

Prajyot01, 2025-08-12:
the Systemsave.json and other files are not generated

closed 2025-09-01 by FASTSHIFT